balena-os/balena-os-device-support #12: Samsung Artik 5

shaunmulligan, 2016-06-20:
**CPU:** ARM based Dual Core @1GHz each
**GPU:** ARM Mali 400 MP2
**Memory:** 512MB LPDDR3
**Storage:** 4GB eMMC
**Features:**
- WiFi
- BT/BLE
- ZigBee/Thread (802.11 b/g/n)
- Secure Element, TEE (TrustZone)

**Link:** https://www.artik.io/hardware/artik-5

bgentry, 2017-05-17:
Do the newer Artik 530 modules work without issue on Resin?

shaunmulligan, 2017-05-17:
@bgentry the 530 is a different SoC, so unfortunately doesn't work out of the box. We plan to add support in the near future though.

emdash, 2017-10-06:
We are trying to get a project off the ground with the 530, a module we chose because we were under the impression that it was already supported (based on both Resin's and Samsung's documentation). Is this still in the pipeline? Is there anything we can do to help move things along?