balena-os/balena-os-device-support #13: Dragonboard 410c

shaunmulligan, 2016-06-20:
**CPU:** Quad-core ARM® Cortex® A53 at up to 1.2 GHz per core with both 32-bit and 64-bit support
**GPU:** Qualcomm Adreno 306 GPU with support for advanced APIs, including OpenGL ES 3.0, OpenCL, DirectX, and content security
**Memory:** 1GB 533MHz LPDDR3 memory
**Storage:** 8GB eMMC 4.51 storage
**Features:**
- Wi-Fi 802.11 b/g/n 2.4GHz, integrated digital core
- Bluetooth 4.1, integrated digital core
- Qualcomm® IZat™ location technology Gen8C
- On-board Wi-Fi, BT and GPS antenna
- HDMI
- 2x USB ports
- Micro SD card slot
- One 40-pin low speed connector
- One 60-pin high speed expansion connector
- Footprint for an optional analog expansion connector for stereo headset/line-out, speakers and analog line-in

**Link:** https://developer.qualcomm.com/hardware/dragonboard-410c

gizero, 2016-10-26:
Hi! Anyone currently working on this? Just got one of these boards and willing to try porting to resin-os, but don't want to duplicate the effort if someone is already doing it

shaunmulligan, 2016-10-28:
Hey @gizero as far as I know, no one has started work on porting the 410c, but I know @imrehg had talked a bit about doing it, perhaps you guys could collaborate?

imrehg, 2016-10-28:
That would be interesting!

gizero, 2016-11-21:
FYI: @PaoloneM. Some initial work on this is happening [here](https://github.com/gizero/resin-dragonboard). This is still in bootstrap phase, but any early comments are highly welcome. Main issues right now:

- still not building any bootloader (seems that upstream bsp layer for the platform doesn't provide any)
- seems like binaries compression with upx tool (used by resin-provisioner recipe) doesn't support aarch64 binaries (?)
- still have to figure out how this beast is booting: main reference I'm going to follow next is [this presentation](https://www.yoctoproject.org/sites/default/files/ypdd-2016.04-san_diego_1.pdf): it should hopefully give hints on how to boot from SD card, hence providing a bootable image.

james10174, 2017-06-11:
Any news on this? Seems like it hasn't been worked on for awhile.