balena-os/balena-os-device-support #18: C.H.I.P Computer

shaunmulligan, 2016-06-20:
**CPU:** 1GHz Allwinner R8
**GPU:** ARM Mali 400 GPU
**Memory:** 512MB
**Storage:**  4GB NAND flash 
**Features:**
- Built in WiFi B/G/N
- Bluetooth 4.0
- Full USB Port
- Composite video port
- Included composite video cable
- Can be powered by battery

**Link:** https://getchip.com/pages/chip

imrehg, 2016-06-30:
> its not possible to run BTRFS on top of unmanaged NAND flash.

Is there any source for this info, or some way to test it? (kernel docs, btrfs wiki, mailing list don't turn up anything that I can find, I think it should be some trivial info somewhere, thanks!)

hh, 2016-08-01:
Do we need to use BTRFS, there are several docker backends https://docs.docker.com/engine/userguide/storagedriver/

AUFS, Device Mapper, and Overlay all seem like valid options and should work with a NAND fs.

agherzan, 2016-08-02:
@hh We are going to switch on overlayfs in the future which will unlock this device. But for now, as in the current state we rely on BTRFS for many of our services, we have to label this board as unsuitable.

hh, 2016-08-02:
@agherzan Can we go ahead and start the layer / build process and override the docker config to use overlayfs? That might allow us to continue working towards support until overlayfs is used across the platform.

Also, what specific services require BTRFS driver on the target device?

agherzan, 2016-08-23:
Sorry @hh . I was on vacation for the past 3 weeks. I don't think it will be a problem for you to check it but please don't use it in production. There are internal discussions and we might choose something else for the official resin OSs.

hh, 2016-09-12:
@agherzan we were planning to start working on the layers again today, did your internal discussions come up with anything firm?

agherzan, 2016-09-13:
Hi @hh. We did indeed decided on this and we are going to switch towards aufs. Actually there is already work in this regards in meta-resin (https://github.com/resin-os/meta-resin/pull/294)  I expect this to come to life pretty soon. We already have a PoC on Beaglebone Black running for testing in house and you can follow up the connected meta-resin issue for progress.

hh, 2016-11-13:
I think we are ready to demo the port of resin-os to CHIP, including our ci pipeline for the build.
We created an org over at https://gitlab.ii.org.nz/iichip to house all the related code

:point_up: [November 1, 2016 9:11 PM](https://gitter.im/resin-os/chat?at=58184e3b49891fe74225f03f)
[image]

hh, 2016-11-13:
New layer for resin on chip:
https://github.com/ii/resin-chip

Had to hack up Resin OS to get around some assumptions (ext4/ubi/etc) C.H.I.P.
https://github.com/resin-os/meta-resin/pull/474

Had to add support for multiple UBI volumes: 
 https://github.com/myfreescalewebpage/meta-chip/pull/1

hh, 2016-11-13:
Build happening now:
https://gitlab.ii.org.nz/iichip/resin-chip/builds/1541

[image]

[image]

hh, 2016-11-13:
Results in working resin deploy that shows up in dashboard!

[image]

hh, 2016-11-14:
https://gitlab.ii.org.nz/iichip/resin-chip/builds/1543 was successfully

imrehg, 2016-11-14:
Hey @hh that looks cool! Got a CHIP here, have to run a build and try it out myself too :)

Links to the builds you mentioned above are 404, I guess it's not public access at the moment?

dlx, 2016-11-14:
Sorry about that, fixed now and should all be Open, 
Build Artifacts are availiable for Download under the ResinOSvCI Branch

imrehg, 2016-11-14:
@dlx cheers, I guess that's in the latest change pushed there, and the build is still ongoing. I'll check back to it a bit later :)

townie, 2016-12-24:
I have a few projects I want to use a chips with resin.io. Do you know when this will be released?

agherzan, 2016-12-24:
Hello @townie. We are working in finding a way forward with this board. Currently we don't have any ETA. All we can say is that is it part of our board priority list.

robeastham, 2017-02-15:
I'm really keen to try one of these boards with resin.io. Would it help speed things up if I hand delivered a crate of beer to the resin.io offices? I only live up the road in London Fields. 

Joking, aside I just wanted to log my +1 here too. Resin support for a board like this, opens up possibilities for product lines that could not be contemplated otherwise.

chacal, 2017-03-26:
+1 This would be really nice!

sashahilton00, 2017-06-22:
Forgive me if I'm wrong, but wasn't support for this sorted a few months back? Saw a gouple of refs to it in gitter, and a link to this repo: https://gitlab.ii.org.nz/iichip/resin-chip which appears to contain a C.H.I.P meta layer?

agherzan, 2017-06-22:
Hi @sashahilton00. Indeed there is work done for chip to work on resin but currently that contains change that forks off resin-os and includes incompatible changes. This is why we restarted this activity and we will integrate with that work so that this board will be fully supported by resin. Hope this clarifies.

kahlkevin, 2018-01-04:
Hi, was just wondering at status. Main https://resinos.io/#features page mentions that resinOS team maintains CHIP BSP layer (see FAQ/"Why Yocto?"). And then there's this project: https://github.com/resin-os/resin-chip.

Are we close to seeing this device 'officially' supported (e.g. with pre-built and updated resin.img downloads, etc.)?

Barring that, if I bake my own image, what is the current level of functionality/compatibility? Issues list?

Thanks for all of the hard work so far, and for any information you can share here!