balena-os/balena-os-device-support #4: Raspberry Pi Zero

shaunmulligan, 2016-06-20:
**CPU**: Broadcom BCM2835 - ARM1176JZFS (ARM11 using an ARMv6-architecture core) with floating point, running at 1Ghz
**GPU:** Broadcom VideoCore IV, OpenGL ES 2.0,OpenVG 1080p30 H.264 high-profile encode/decode, 250 MHz
**Memory:** 512MB RAM
**Features:** 
- USB On-The-Go port
- Micro USB power
- micro HDMI
- Composite Video

**Link:** https://www.raspberrypi.org/products/pi-zero/

joncutrer, 2017-03-14:
Please add support for new Pi Zero W, onboard wifi variant.

cjdcordeiro, 2017-08-18:
Hi,

is this already supported or not? I've seen #220 is closed but I've just tried it on my RPI 0 W, and nothing happens...the green led just keeps blinking and no screen. On the RPi 3 it works though...

telphan, 2017-08-18:
This is already supported. Just a note, there are different images for rpi3 and rpi0.

cjdcordeiro, 2017-08-18:
Thank you. I've noticed that rpi1 is the respective version.

gardere, 2019-02-04:
I noticed that RPi 1 and RPi 0 are not available when trying to "add a device" from within an application dashboard. However it is possible to download an OS version that suits these devices from balena.io landing page (https://www.balena.io/os/#download)

[image]

[image]