balena-os/balena-os-device-support #9: Beaglebone Green Wireless (BBGW)

shaunmulligan, 2016-06-20:
**CPU:** Sitara AM3358BZCZ100: 1GHz, 2000 MIPS
**GPU:** NEON: SGX530 3D, 20M Polygons/S
**Memory:** 512MB DDR3 RAM
**Storage:** 4GB 8-bit eMMC on-board flash storage
**Features:**
- 2x PRU 32-bit microcontrollers
- USB client for power & communications
- USB host with 4-port hub
- WiFi 802.11 b/g/n 2.4GHz
- Bluetooth 4.1 with BLE
- 2x 46 pin headers
- 2x Grove connectors (I2C and UART)
**Link:** http://beagleboard.org/green-wireless

xoul, 2017-06-01:
I have just flashed two of them with Resin 1.26.0.
They seems to have the same IP address, because of a commond "deadbeef" MAC address on the WiFi.

shaunmulligan, 2017-06-01:
@xoul this has been fixed in the resinOS 2.0, you should be able to just use a "Beaglebone Black" image instead of green, as the 2.0.4 version of Beaglebone black will work on all beaglebone variants

xoul, 2017-06-02:
I'm still running one of my BBGW with the recommended 1.26.0 and I've tried both 2.0.0-rc1.rev1 and 2.0.0-rc1.rev1-dev with the other, but this does not even seem to get an IP address (while the first one is powered off, so no conflict on the Deadbeff MAC address).
I'll wait for the 2.0.4 release.

shaunmulligan, 2017-06-02:
@xoul please re-read my previous message. You can use the Beaglebone **Black** image on the green devices, we are going to consolidate all of these types into one device type. so the BBGW applications are not going to be updated, as the recommended way forward is to use BBB applications for all beaglebone variants.

xoul, 2017-06-02:
If I re-read your previous message I see that you were talking about 2.0.4 and I this is not available on the main Resin site right now. ;) You were also saying that ResinOS 2.0 fixes the problem, but 2.0.0 BBGW-related does not seem to fix the problem right now.
I'm going to try 2.0.3 for BBB on BBGW now.

shaunmulligan, 2017-06-05:
Indeed, sorry about that, I meant 2.0.3 (or whatever is the latest) for the BBB builds. Sorry for the confusion.