balena-os/balena-os-device-support #26: Odroid C2

merlinthp, 2016-08-03:
**CPU:** Amlogic ARM® Cortex®-A53(ARMv8) 2Ghz quad core CPUs
**GPU:** Mali™-450 GPU (3 Pixel-processors + 2 Vertex shader processors)
**Memory:** 2Gbyte DDR3 SDRAM
**Features:**
- Gigabit Ethernet
- HDMI 2.0 4L/60Hz display
- H.265 4K/60FPS and H.264 4K/30FPS capable VPU
- 40pin GPIOs + 7pin I2S
- eMMC5.0 HS400 Flash Storage slot / UHS-1 SDR50 MicroSD Card slot
- USB 2.0 Host x 4, USB OTG x 1 (power + data capable)
- Infrared(IR) Receiver

**Link:** [http://www.hardkernel.com/main/products/prdt_info.php?g_code=G145457216438](url)

merlinthp, 2016-08-03:
I'm not spotting any ARMv8 devices on your supported list right now, so I realise this'll not be an entirely trivial board to support.

tomarthur, 2017-09-01:
Is there an updated ETA for Odroid-C2 support?

shaunmulligan, 2017-09-01:
Hey @tomarthur unfortunately there isn't anyone working on this yet and we haven't seen it requested very much. I think it shouldn't be too difficult to support, so if you have a bit of yocto knowledge or are will to learn some, you might start looking at adding it to https://github.com/resin-os/resin-odroid, and our team is there on https://gitter.im/resin-os/chat to help

d-simon, 2017-09-01:
I'd be super happy to have this 👍🏻
(we haven't got around to attempt our own image.)

mibstar, 2019-03-05:
I'm also looking into an set-top-box project and while 4K is not a requirement smooth media playback at 1080p is. The AMLogic boards are very popular for this use-case. 

https://en.wikipedia.org/wiki/Amlogic#Media_player_SoCs_(S8_family)