balena-os/balena-os-device-support #27: Beaglebone Green

leite08, 2016-08-08:


leite08, 2016-08-08:
https://beagleboard.org/green

shaunmulligan, 2016-08-08:
@leite08 The beaglebone green is already supported and should work out the box if you use a standard beaglebone image. We also support the Beaglebone Green Wireless, which you will see as `Beaglebone Green Wireless (BETA)` in the drop down.

Have you had issues with provisioning a BBG?

leite08, 2016-08-09:
Hey Shaun, thanks for the reply!

Yes, I've tried to provision a BBG with 3 different SDCards we use to setup
our BBBs and it didn't work.

Will try again with a freshly burned one and will report here later. There
is a possibility we've bricked them when fixing the /ttyS5 HDMI to Serial
config...

Cheers

Rafael Leite

leite08, 2016-08-10:
Hey!

I've tried to flash the BBG with a newly downloaded app image and it didn't
work.

Then I removed the ​SDCard, inserted it on a BBB and it worked.

The steps and what happened:

   1. As usual, inserted te SDCard in the offline BBG;
   2. Pressing the S2 button for 10 seconds, powered it up;
   3. The user leds started blinking as usual when provisioning a new
   device;
   4. The dashboard of Resin.io showed the new device, with the status
   "Configuring";
   5. The device's status bar went quickly from 0 to 100% (quick but
   acceptable - didn't seem an error);
   6. The device got stuck in the "Configuring" status, user leds blinking
   and it didn't move to the "Post provisioning" state, as it happened with
   the BBB.
   7. Turned it off, removed the SDCard and powered it up again, it didn't
   appear online on Resin.io's dashboard anymore.

Any clues or suggestions?

Rafael Leite

agherzan, 2016-08-11:
@telphan can You please take a look on this?

telphan, 2016-08-11:
@leite08 It seems there is an issue with the HDMI in our new images which renders our BBG image problematic. We are investigating the issue to come up with a solution. Will keep you posted.

leite08, 2016-08-11:
Thanks Theodor!

There is a known issue with the /ttyS5 being used for HDMI instead of
serial, but it doesn't impact in provisioning the BBB.

Cheers!

Rafael Leite

telphan, 2016-08-11:
@leite08 The issue is related to using the wrong dtb on BBG (the same as BBB) I can modify an image for you to load the proper dtb and this will unblock your device, but for this please open a support ticket and I will provide you with everything you need.