balena-os/balena-os-device-support #28: Up Board

imrehg, 2016-08-24:
* **CPU**:  Intel® Atom™ x5-Z8350 Processor (2M Cache, up to 1.92 GHz) CPU with 64 bit architecture; Quad Core CPU
* **GPU**: Intel® HD 400 Graphics
* **Memory**: 1GB/2GB/4GB DDR3L-160
* **Storage**: 16GB / 32 GB / 64 GB eMMC
* **Features**:
 * Gigabit Ethernet
 * 4xUSB2.0
 * 1xUSB-OTG
 * 40-pin GP-bus
 * CSI camera interface
 * HDMI / DSI / eDP

* **Link**: http://www.up-board.org/en/specifications/

mevatron, 2016-11-22:
Any progress on this port? I have an UP Board in house now, and I would be happy to help.

I also found emutex has already developed a [Krogoth Yocto BSP](https://github.com/emutex/meta-up-board) for the UP board so that might be helpful for some things.

agherzan, 2016-11-25:
Hello @mevatron . Thanks for your interest in the resin platform. We don't have this as a high priority right now but we do encourage people to try our board support document and contribute in getting new boards supported. https://github.com/resin-os/meta-resin/blob/master/contributing-device-support.md . We will be around for any help/support you might need.

curcuz, 2017-02-26:
initial suppport added in https://github.com/resin-os/resin-up-board/pull/1

agherzan, 2017-02-26:
Available only in staging for now.

imrehg, 2017-03-15:
Now supported: https://resin.io/blog/up-board-support/