balena-os/balena-os-device-support #32: A38X Clearfog Pro - Solidrun

Valks, 2016-09-26:
MicroSOM: A388 SoM 
CPU: Dual core ARM Cortex-A9 up to 1.6GHz
GPU: N/A
Memory: 32-bit DDR3L 1GB (up to 2GB)
Features:
2 x mSATA/mPCIE
1 x USB 3.0 port
1 x Port dedicated Ethernet
6 x Port switched Ethernet with jumbo frame (10KB)
1 x SFP

Analog Audio/TDM module support
GPIO Header (MikroBus)
Indication LEDs
User Push Buttons
PoE expansion header
RTC Battery
FTDI (Console Only) / Debug Header
JTAG Header 

Storage
M.2
uSD / eMMC (optional)

Link: [http://www.solid-run.com/wiki/doku.php?id=products:a38x:clearfog](http://www.solid-run.com/wiki/doku.php?id=products:a38x:clearfog)

linux4kix, 2017-11-24:
Hi,  Jon here from SolidRun.  I just caught up with your request and can provide you with a meta-layer for testing if you are still interested.  One of the main points I am still working out is how to manage the various ports for connectivity.  It is easy to specify an ethernet, wifi, or cellular connection to be managed by the system to provide Resin.io connectivity.  The challenge comes in then managing the other interfaces from within the docker container.

Valks, 2017-11-29:
Hi Jon, still interested. My main project for this has already shipped, but I'd still like to see it working on resin as I feel it would provide better long term management at the OS level.

linux4kix, 2017-12-01:
Great!  I will pull together some instructions for building it and then post the repositories and pull requests here.

Valks, 2018-05-24:
@linux4kix Hi Jon, thought I'd follow up and see if you have the meta-layer available? I'm still keen to try it and see if I can use it in place of my custom OS setup I currently use.

linux4kix, 2018-05-24:
Yes,  sorry about the delays.  Your rocko meta-layer that is compatible
with Resin.io is just about ready for release.  I have been reworking our
4.9 and 4.14 kernels with the latest patchsets and u-boot releases.  U-Boot
2018.01 has been released and the kernel should be finishing internal QA
today or tomorrow and then the meta-layer will follow.

Valks, 2018-05-24:
Very nice timing. Should line up well with my plans for testing.

Valks, 2018-06-06:
@linux4kix Any update on this issue? If it's ready to go I'd like to give it a try.

jnettlet, 2018-06-11:
Getting pushed tonight.  @Valks after you test it I will get a pull request ready for ResinOS.

Valks, 2018-06-15:
@jnettlet Any follow up on this issue?

jnettlet, 2018-06-17:
@Valks, yep.  Ended up traveling which has held me up.  As soon as I land at the office in Israel I will push the repo.  Initial testing has been stable on the cf-pro and base

Valks, 2018-06-28:
@jnettlet any chance you can push the repo?