balena-os/balena-os-device-support #35: Beagleboard X15

Neuromancer2701, 2016-10-12:
* **CPU**:  Sitara AM5728 with dual-core 1.5GHz Cortex-A15 processor,
* **GPU**: 532MHz SGX544 3D graphics processor,
* **Memory**: 2GB DDR3L SDRAM
* **Storage**: 4GB eMMC Flash memory
* **Features**:
- dual 700MHz C66x DSPs and dual 212MHz Cortex-M4 image processors
- 3 x USB 3.0 Host connectors
- USB 2.0 Client on microUSB connector
- Powered eSATA connector
- 2 x 10/100/1000 Ethernet RJ45 connectors
- HDMI video and audio connector
- 3.5mm jack sockets for audio in/out
- JTAG serial debug header
- microSD card slot
- 4 x 60-pin Expansion connectors
- 157 x GPIO pins
- 185 x Programmable Real Time Unit (PRU) pins
- 4 x USB 2.0 Host ports
- 2 x LCD display ports
- 2 x PCIe channels
- Camera port
- UART, SPI, I2C, CAN serial ports
 
* **Link**:http://elinux.org/Beagleboard:BeagleBoard-X15

jturnsek, 2017-01-30:
Am using resin-571x-evm for the TMDSEVM572X board, which is actually almost the same as Beagleboard X15 and 
it works, except bootloader should be placed on eMMC. instead of Serial Flash. See issue https://github.com/resin-os/resinos/issues/164.

agherzan, 2017-01-30:
This sounds like a new machine type. CC @floion