balena-os/balena-os-device-support #37: Pine64

willrea, 2016-11-03:
* **CPU**:  Allwinner Quad-core Cortez-A53 64bit @ 1.15Ghz
* **GPU**: ARM Mali400MP2
* **Memory**: 512MB/1GB/2GB DDR3
* **Storage**: microSD up to 256GB

* **Features**:
DSI - Display Serial Interface, 4 lanes MiPi, up to 1080P
CSI - CMOS Camera Interface up to 5 mega pixel
TP - Touch Panel Port, SPI with interrupt
RTC - Real Time Clock Battery Connector
Wifi/BT Module Header - SDIO 3.0 and UART
2x20 pins "Pi2" GPIO Header
2x17 pins "Euler" GPIO Header
2x5 pins "EXP" Console Header
Gigabit Ethernet port

* **Link**: http://pine64.org

abelboldu, 2017-01-30:
https://github.com/mont3z/meta-pine64

elijah, 2017-01-31:
I have a pine64 board but really don't know enough about the resin build process to bolt the meta-pine64 hardware support bits into everything else.   I support the notion of someone doing this work, though.  ;-)

Zeustopher, 2017-04-20:
This would finally be the thing that lets me use my Pine64 to its potential!

sent-hil, 2018-01-21:
👍

dirkmoors, 2018-03-20:
Any update on this platform?

dukuo, 2018-08-13:
What's the status on this board? Anything we can do to speed up the process ?

krital, 2018-10-06:
Can we provide any assistance to get this moving?

gcamerli, 2018-11-21:
Hi all, I opened a new issue for the model RockPro64. Check out #70