balena-os/balena-os-device-support #38: NanoPi NEO Air

imrehg, 2016-11-09:
* **CPU**:  Allwinner H3, Quad-core Cortex-A7 Up to 1.2GHz
* **Memory**: 512MB DDR3 RAM
* **Storage**: 8GB eMMC + uSD
* **Features**:
  * 802.11b/g/n
  * Bluetooth 4.0
  * GPIO

* **Link**:
  * http://nanopi.io/nanopi-neo-air.html
  * http://linux-sunxi.org/FriendlyARM_NanoPi_NEO
 
Forum request: https://talk.resin.io/t/support-for-nanopi-neo-air/381

cara, 2017-01-14:
Would love to have this, too!

unixbigot, 2017-03-07:
+1 for this.  I have run docker on the NanoPi Air using the Armbian nightly builds (4.10 kernel) and this board is a little gem.

shaunmulligan, 2017-07-28:
Hey @cara and @unixbigot , in my spare time I have been building support for the nanopi and I know have images that are is a good enough state for people to test and play around with them. So if that is something the two of you are interested in, just let me know. 

There are still a few little things to get right before this will be available for download in the resin.io platform, mainly I need to sort out the bootloader integration with the resin hostOS upgrade mechanism.

unixbigot, 2017-07-29:
Yep, I have a couple of idle boards I can use to test.

—chris

— 
Christopher Biggs      <christopher@biggs.id.au <mailto:christopher@biggs.id.au>>
Brisbane, Australia.   +61-407-722-799  @unixbigot

shaunmulligan, 2017-08-04:
Alright. So I finally got this all up for everyone to test. Here are the instructions:

## Provisioning a Nano Air
### ++++++++++++++++++++++++++++++++++++++++++++++
### Warning: this is still very alpha level support, so you shouldn’t use these images for anything serious. 
### ++++++++++++++++++++++++++++++++++++++++++++++

1. Create a raspberry pi 3 application on resin.io Dashboard, call it something like “nanopi”
2. Download the config file for the application. this can be found in the “advanced” section after clicking “DOWNLOAD RESINOS”. You should get a file called `nanopi.config.json`
3. Download either `resin-nanopi-emmc.img.zip` or `resin-nanopi-sdcard.img.zip` from https://drive.google.com/open?id=0B1bWigECStTrQUhlY0xiSUd5Nk0, extract the `.img` from the compressed file. The emmc image will install itself on to the nanopi’s internal emmc storage, so bewared as this will erase any other data you have there.
4. mount the extracted `.img` and you want to mount the `resin-boot` partition. 
5. Copy the `nanopi.config.json` to the boot partition and rename it to `config.json` . You will probably have to delete the current config.json first.
6. setup the network config. In the `resin-boot` partition you should find a folder called `system-connections` and in there a connection file called `resin-sample.ignore`. Rename this file to something like `resin-wifi` and also edit the `ssid` and `psk` values in that file to match those for your wifi router so the device can connect.
7. unmount the `.img` and flash it onto an SD card.
8. pop the SD card into the nanopi and boot it up. You should see the nanopi pop up in the application you created in step 1.

## Notes
* Both builds are `.dev` builds, which means you can ssh into them using `ssh root@<DEVICE_IP> -p22222`
* To load specific hardware interfaces, the OS uses the same mechanism as Armbian, you can see that described here: https://docs.armbian.com/Hardware_Allwinner_overlays/
* By default the following DT overlays are enabled: `uart1 i2c0 spi-spidev analog-codec`, so you should have access to i2c, uart, spi and audio output.
* The following things are known to not be working:
    * resin sync
    * camera interface
    * TV out interface
    * I2S audio interface
    * HDMI interface (nanopi air doesn’t have this anyways) 

cc @imrehg , @unixbigot and @cara let me know what you think, or if you hit any issues.

shaunmulligan, 2017-08-04:
Just a note, if your Nanopi Air has a `AP6212a` wifi chip, chances are it won't connect to the wifi :cry:

splitice, 2017-10-19:
@shaunmulligan I too have need for this. Could you perhaps share some information on how this build was made?

We have units with the both the new and earlier WiFi chip. Armbian has support now days. We should be able to utilize their overlays right?

shaunmulligan, 2018-01-27:
@splitice sorry, some how i missed this comment. The build was made from here: https://github.com/shaunmulligan/resin-allwinner . I am hoping to finally get this moved over to resin-os officially and have it supported in the dashboard. The only thing outstanding is the uboot adaptions for hostOS upgrades: https://github.com/shaunmulligan/resin-allwinner/issues/9

Also, yes the armbian overlays, I have test some of them, but not all yet.

shawaj, 2018-10-22:
@shaunmulligan will this work ok with a Nano Pi Neo Core / Core2 do you think?

CC @ryanteck

shaunmulligan, 2018-10-22:
@shawaj probably not as the dtb is different, but it should be fairly easy to add support for that board in the resin-os/resin-allwinner repo

splitice, 2018-11-30:
We have been running a NEO Air build from our branch for over a year now. It's open source but you will need to build your own images.

https://github.com/HalleyAssist/resin-allwinner/tree/the-branch-that-builds

Note: We were not successful in upgrading our Yocto version (theres seemingly a yocto bug that's been holding us back) so are still building on an older kernel & supporting utils (e.g systemd).

shaunmulligan, 2018-11-30:
@smomni and @splitice its actually in our staging environment currently ( https://dashboard.balena-staging.com ) if you want to play around and test it. It hasn't passed our checks yet. There seems to be an issue with rebooting but the device team hasn't had time yet to investigate. Once that is fixed it should make its way to production.

splitice, 2018-12-03:
@shaunmulligan Do you mean an issue with the device randomly rebooting, or an issue with the device when performing a reboot.

There are three fixes in we have for random crashes (usually reboots) in our repositories:
1.  Enable IO counting in the scheduler. This prevents stalls under low CPU load & high IO load (applies to any MMC/SD based device). We currently do this in our application.
2. Setting the dtb thermal thresholds to reasonable value (default max is 125 degrees!).
3. Extending the systemd watchdog timer to 16s due to bugs in systemd that can cause blocking (this should not be necessary on new yocto).

I'm happy to supply code for either of these.

FYI @brdrcol

smomni, 2018-12-14:
@shaunmulligan thanks for the info! I tried to add a new device to my application and download the balenaOS image from https://dashboard.balena-staging.com but after clicking the download button ("Download balenaOS(~NaN  undefined)" something funny going on there) the redirect to https://api.balena-staging.com/download returns a 502 Bad Gateway error. Any ideas what's the issue here?

adamtec, 2019-03-13:
I have successfully got a NanoPi NEO Air up and running with BalenaOS v2.26.0. It appears there is currently no support for g_ether for USB OTG? Perhaps, I am overlooking something but I do not see an easy way to enable this. Any thoughts?

imrehg, 2019-03-14:
@adamtec opened an issue for this in the relevant repository, where you can follow any updates regarding that: https://github.com/balena-os/balena-allwinner/issues/70