balena-os/balena-os-device-support #40: Orange pi zero

lcasassa, 2016-12-16:
Hardware specification

CPU
H2 Quad-core Cortex-A7 H.265/HEVC 1080P.

GPU
·Mali400MP2 GPU @600MHz
·Supports OpenGL ES 2.0

Memory (SDRAM)
256MB/512MB DDR3 SDRAM(Share with GPU)(256MB version is Standard version)

Onboard Storage
TF card (Max. 64GB)/ NOR Flash(2MB Default not posted)

Onboard Network
10/100M Ethernet RJ45 POE is default off.

Onboard WIFI
XR819, IEEE 802.11 b/g/n

Audio Input
MIC

Video Outputs
Supports external board via 13pins

Power Source
USB OTG can supply power
USB 2.0 Ports

Only One USB 2.0 HOST, one USB 2.0 OTG

Buttons
Power Button

Low-level peripherals
26 Pins Header, compatible with Raspberry Pi B+

13 Pins Header, with 2x USB, IR pin, AUDIO(MIC, AV)

LED
Power led & Status led

Supported OS
Android, Lubuntu, Debian, Raspbian

Link: http://www.orangepi.org/orangepizero/

lcasassa, 2016-12-16:
I have to add that it costs only 6.99 usd without sd and without shipping.

cara, 2017-01-11:
What love to have this, too!

nmaas87, 2017-01-24:
Would be awesome 👍

matthewepler, 2017-02-22:
+1 for this. The built-in wifi + ethernet + USB dongle for cellular makes it an easy choice for setting up a device with fail-safe connectivity.

unixbigot, 2017-03-07:
+1 for this, these boards are very well supported by armbian nightly builds with the 4.10 mainline kernel, and run docker fine.  (The "legacy" kernel provided by the manufacturer is too old to run docker).

Building on armbian to add support for the orange pi zero should open up the whole orange pi range, as the difference between each is basically mixing and matching peripherals.

bandrews, 2018-05-14:
Adding my voice to the chorus - this is one of the best price/performance tiny Linux boards out there, and would unlock a whole new batch of applications for Resin OS.

battuashwik, 2018-11-29:
Any update on this?

mwohlert, 2019-02-17:
Should be included once my pr is merged for Orange Pi Zero support.
balena-os/balena-allwinner#65

floion, 2019-03-05:
This has now been contributed and will appear in the dashboard soon

joy4eg, 2019-03-22:
Hey,

Any news when this will be available ? :)

floion, 2019-03-22:
Working to make this available sometime next week

floion, 2019-03-23:
This is now available in production. Thanks for the contribution!
And enjoy!