balena-os/balena-os-device-support #41: C.H.I.P. Pro

cara, 2017-01-20:
* **CPU**: 1GHz Allwinner R8 ARM Cortex-A8 processor
* **GPU**: Mali400 GPU
* **Memory**: 256MB Nanya DDR3 DRAM
* **Storage**: 512 MB SLC NAND
* **Features**:
 *Built in WiFi B/G/N
 *Bluetooth 4.2
 *Can be powered by battery
 *I2S Digital Audio	

* **Link**: https://getchip.com/pages/chippro

Comparison from C.H.I.P to C.H.I.P Pro:
https://docs.getchip.com/chip_pro.html#c-h-i-p-v-c-h-i-p-pro

robeastham, 2017-02-15:
Keen to hear about C.H.I.P pro support to.
+1

cara, 2017-02-24:
Especially because C.H.I.P Pro is also intended for mass production and not just makers, it could be a big enrichment for resin

KemperVreden, 2017-03-28:
Seriously - 256 MB RAM and just 512 MB Storage - if anyone manages it to get resinOS running on these - please let me know. However I am pretty sure that resinOS won't be able to fit onto such a small storage!?

cara, 2017-03-28:
Had the 4GB of the C.H.I.P in mind... but you're right it's just 512MB. I think the only way it is going to work if you're going to boot form a sd card.. (with an extern reader)  
By the way, good and interesting answer of simon kemper concerning this topic: https://forums.resin.io/t/non-usb-based-wi-fi-support-for-raspberry-pi-zero/498/15

mukulseth, 2017-05-01:
I would like to see support added too.. is there an image that will work with the CHIP Pro as is? If not with the onboard storage, maybe with an SD card reader? Has anyone tried the generic armv7 debian image?