balena-os/balena-os-device-support #42: ASUS Tinker Board

imrehg, 2017-01-23:
* **CPU**: Quad core 1.8GHz ARM Cortex-A17 CPU
* **GPU**:
* **Memory**: 2GB Dual channel LPDDR3 memory
* **Storage**: micro-SD
* **Features**:
 * Gigabit LAN and Bluetooth 4.0 + EDR connectivity
 * 802.11 b/g/n Wi-Fi
 * 4x USB 2.0 ports
 * 40-pin Internal header with 28 GPIO pins
 * Contact points for PWM and S/PDIF signals
 * 1x 3.5mm Audio jack connection
 * CSI port for camera connection
 * DSI port supporting HD resolution
 * 1x HDMI 2.0 port to support 4K resolution

* **Link**: http://cpc.farnell.com/asus/90mb0qy1-m0eay0/tinker-board-2gb-1-8ghz-4k-gb/dp/SC14363

xanview, 2017-07-16:
Anyone managed to get ResinOS running on the Asus Tinkerboard?

It's a really great board and makes a really great alternative for Raspberry Pi: significantly faster and double the ram, power management, audio input, etc. Having the same factor, it can also be used with the same enclosures as the Pi.

xanview, 2017-07-18:
This seems relevant: https://github.com/sat8/yocto-tinkerboard

shaunmulligan, 2017-07-18:
Hey @xanview ! yeah the Tinkerboard is a great little board. I was actually looking into how difficult it would be to support, it looks to be pretty straightforward. I think the preferred yocto BSP layer is this one: https://github.com/rockchip-linux/meta-rockchip which will most likely also support the [rock64](https://www.pine64.org/?page_id=7147) as well. 

I don't think anyone on the team currently has one of the Asus boards yet, but we will definitely be ordering a few :)

hecontreraso, 2017-07-19:
I'm also very interested in make it resin + tinker board (:

hecontreraso, 2017-07-31:
@xanview did you have any luck running resin OS on the board?

anthify, 2017-08-09:
I would love to get this working with resin and running electron/chromium. Any advice on the necessary steps? I'm not very experienced with this kind of task, I largely work with frontend web technologies.

imrehg, 2017-08-09:
Hi @anthify, yeah, device support is a bit more involved, but trying to make it easier by the day. Here's the documentation for adding custom board support https://resinos.io/docs/contrib/#support-for-additional-boards

floion, 2019-03-05:
Device is supported for a while, closing issue

closed 2019-03-05 by floion

imrehg, 2019-03-07:
@floion @shaunmulligan do we close these when the device type is supported? I thought we are just adding a flag for being supported, as it's easier to find. But closing makes sense, and might need to review the whole list again, if that's what we decide to do from this on?

floion, 2019-03-07:
Yeah, I started closing them because it's way too many open and thought it was easier to close. But if we want to just add the label, we can do that too