balena-os/balena-os-device-support #45: NVidia TX1

jbcpollak, 2017-02-24:
It would be great if Resio.io supported the TX1:

* **CPU**:  64-bit ARM® A57 CPUs
* **GPU**: 256-core NVIDIA
* **Memory**: 4GB
* **Storage**: 16GB eMMC, also SATA interface

* **Link**: http://www.nvidia.com/object/jetson-tx1-module.html

mr337, 2017-05-17:
TX2 is now released also, https://developer.nvidia.com/embedded/buy/jetson-tx2.

mr337, 2017-05-17:
Also looks like there is a BSP layer, not 100% sure if its official. 
https://github.com/madisongh/meta-tegra

I have hardware of TX1 and TX2.

shaunmulligan, 2017-05-17:
@jbcpollak and @mr337 we actually just started work on supporting the TX1 and the TX2 will follow shortly after that, so keep an eye out for it in the next month or so.

kgreenek, 2017-06-30:
Any updates on this? I've been following this thread eagerly. At my company, our platform is based on the TX1 (soon to move to the TX2). We would LOVE to be able to deploy using resin / docker. We still need to be able to leverage the full power of the GPU though.

mr337, 2017-06-30:
Hope its coming down soon. I too need full access to GPU (one of the reason to go TX1/TX2). From all my previous conversations the entire hardware is accessible from inside a container.

jd20, 2018-02-23:
I see TX2 is now supported, is the TX1 still on the roadmap too?

mr337, 2018-02-28:
@jd20 My hunch is if they haven't supported TX1 yet, they probably won't since it is yesterdays technology. I am hoping that I am wrong, but just my two cents.

Aside from cost to upgrade, is there anything preventing to upgrade to the TX2? It is well worth the upgrade IMO.

telphan, 2018-03-02:
@mr337 @jd20 We do plan on releasing the TX1, unfortunately the TX2 work revealed a few problems with the tegra boards which blocked the release of the TX1. But we have a complete PR now to support it and should we released shortly.
https://github.com/resin-os/resin-jetson-tx1/pull/1

jd20, 2018-03-02:
@telphan Cool, that's great to hear.

alisondavis17, 2018-05-15:
Hi all, we've just released beta support for the TX1!

telphan, 2018-06-17:
This is fully supported now. Closing the ticket.

closed 2018-06-17 by telphan