balena-os/balena-os-device-support #47: BeagleBoneBlue

imrehg, 2017-03-21:
* **CPU**:  Octavo Systems OSD3358 1GHz ARM® Cortex-A8
* **Memory**: 512MB DDR3 RAM
* **Storage**: On-board 4GB 8-bit eMMC / SD Card Slot
* **Features**:
  * Wireless: 802.11bgn, Bluetooth 4.1 and BLE
  * 2×32-bit 200-MHz programmable real-time units (PRUs)
  * ARM Cortex-M3
  * Motor control: 8 6V servo out, 4 DC motor out, 4 quadrature encoder in
  * Sensors: 9 axis IMU, barometer
  * Connectivity: HighSpeed USB 2.0 client and host
  * User interface: 11 user programmable LEDs, 2 user programmable buttons

* **Link**: http://beagleboard.org/blue

imrehg, 2017-04-11:
In progress: https://github.com/resin-os/resin-beaglebone/pull/79

willnewton, 2017-04-20:
This board is now supported.