balena-os/balena-os-device-support #50: BananaPi M2+

tzahari, 2017-04-10:
* **CPU**:  H3 Quad-core Cortex-A7 H.265/HEVC 4K
* **GPU**: Mali400MP2 GPU @600MHz,Supports OpenGL ES 2.0
* **Memory**: 1GB DDR3 (shared with GPU)
* **Storage**: TF card (Max. 64GB) / MMC card slot,up to 2T on 2.5 SATA disk, 8GB EMMC Flash
* **Features**: 
  * Onboard Network: 10/100/1000M Ethernet RJ45
  * Onboard WIFI: SDIO AP6212（option AP6181, AP6335
  * USB 2.0 Ports: two USB 2.0 HOST, one USB 2.0 OTG
  * Video Input: A CSI input connector Camera:1 Supports 8-bit YUV422 CMOS sensor interface,2 Supports CCIR656 protocol for NTSC and PAL,3 Supports SM pixel camera sensor ,4 Supports video capture solution up to 1080p@30fps
  * Video Outputs: Supports HDMI output with HDCP,Supports HDMI CEC, Supports HDMI 30 function,Integrated CVBS, Supports simultaneous output of HDMI and CVBS
  * Audio Output: HDMI
  * Low-level peripherals: 40 Pins Header,compatible with Raspberry Pi B+
  * uart GPIO(1x3) pin: UART, ground
  * Supported OS: Android, Ubuntu, Debian, Rasberry Pi Image

* **Link**: https://bananapi.gitbooks.io/bpi-m2-/content/en/bpim2+hardwarespec.html

tzahari, 2017-04-14:
I have the hardware but I can't add the label?

alisondavis17, 2018-05-15:
Just a note that the BananaPi M1+ is now supported (but not the M2+ yet)