balena-os/balena-os-device-support #51: Yantrr VIBE2

xoul, 2017-05-31:
* **CPU**:  TI Sitara ARM Cortex-A8 AM335x , up to 1GHz, 2000MIPS
* **GPU**: 
* **Memory**: 512 MB
* **Storage**:  4GB eMMC + microSD
* **Features**:
  * Fully BeagleBone compliant
  * 2G/3G/HSPA+ with world-wide support
  * 10/100 Ethernet
  * Optional WiFi 802.11 b/g/n
  * Optional GPS with Timing Synchronization
  * Optional Smart Bluetooth 4.1 (BLE)
  * Optional HDMI with micro connector
  * Optional automatic battery backup
  * Optional battery-backed precision Real Time Clock

* **Link**: http://yantrr.com/vibe2

VIBE2 is actually a BBB on steroids.
Also note that @lifeeth uses Yantrr boards as seen on https://forums.resin.io/t/cellular-modem-support/195/5!

shaunmulligan, 2017-05-31:
hey @xoul , I actually have one of the older yantrr boards. The VIBE2 should just work out of the box using the standard beaglebone Black image, have you given it a try?

xoul, 2017-06-02:
Yes, I've tried the BBB image on my VIBE2 before, but I've not deeply tested support for 3G, GPS, WiFi and Bluetooth.
Morevoer the Yantrr support team says that this would be mandatory to operate correctly a VIBE2:

> For VIBE2, very first you need to change the USB0 dr_mode from peripheral to host in Beglebone Black kernel source file KERNEL/arch/arm/boot/dts/am335x-bone-common.dtsi while rebuilding the kernel
> &usb0 {
>         status = "okay";
>         dr_mode = "host";
> };