balena-os/balena-os-device-support #52: Hilscher netPI - ruggedized Raspberry Pi 3 with Industrial network controller

hilschernetpi, 2017-06-05:
* **CPU**:  BCM 2837, the same as Pi 3
* **GPU**: BCM 2837, the same as Pi 3
* **Memory**: 1 GByte 
* **Storage**: 8 GByte default (micro SDcard)
* **Features**: Pi 3 circuity combined with Hilschers automation controller [netX](https://www.hilscher.com/netx/introduction/) made for Industrial Networks such as Profinet, EtherNet/IP, EtherCAT and many more
  * 125Mhz SPI high-speed connection between BCM and netX
  * 8 KByte FRAM with nearly endless write cycles to store data non-volatile frequently
  * Maintenance free hardware RTC with supercapacitor backup buffering
  * Special BCM2837 cooling to support full CPU quad-core speed @1.2GHz up to 50°C
  * Temperature range -20°C to +60°C
  * Expansion slot for additional network modules
  * Hardened, AppAmor secured Yoco Linux Kernel 4.4
  * Pre-installed Docker deamon
  * [portainer.io](http://portainer.io/) web GUI frontend to handle Docker images and containers
  * Release Q3/2017
* **Link**: [element14 blog](https://www.element14.com/community/community/raspberry-pi/blog/2017/03/28/a-new-ruggedized-raspberry-pi-3-netpi)

shaunmulligan, 2017-06-05:
Does the netX hat have a custom dtb or is it included in the vanilla raspbian builds? if it is, then we should just get support for that hat and this board for free when we pull the dtb in.

hilschernetpi, 2017-06-05:
gfidisc.hilscher.com {color:black; !important} 
  

Hi Shaun,

no it is not. Today we offer just a library *.a for netX to be included when your compile an application using GCC.

So in general is that something you would recommend to us to make a dtb? 

Thx
Armin

 

view it on GitHub, or 
mute the thread.

Hilscher Gesellschaft für Systemautomation mbH   |  Rheinstrasse 15  |  65795 Hattersheim  |  Germany  |  www.hilscher.com Sitz der Gesellschaft / place of business: Hattersheim  |  Geschäftsführer / managing director: Dipl.-Ing. Hans-Jürgen HilscherHandelsregister / commercial register: Frankfurt B 26873  |  Ust. Idnr. / VAT No.: DE113852715Registergericht / register court: Amtsgericht Frankfurt/MainImportant Information:This e-mail message including its attachments contains confidential and legally protected information solely intended for the addressee. If you are not the intended addressee of this message, please contact the addresser immediately and delete this message including its attachments. The unauthorized dissemination, copying and change of this e-mail are strictly forbidden. The addresser shall not be liable for the content of such changed e-mails. Wichtiger Hinweis: Diese E-Mail einschließlich ihrer Anhänge enthält vertrauliche und rechtlich geschützte Informationen, die nur für den Adressaten bestimmt sind. Sollten Sie nicht der bezeichnete Adressat sein, so teilen Sie dies bitte dem Absender umgehend mit und löschen Sie diese Nachricht und ihre Anhänge. Die unbefugte Weitergabe, das Anfertigen von Kopien und jede Veränderung der E-Mail ist untersagt. Der Absender haftet nicht für Inhalte von veränderten E-Mails.

shaunmulligan, 2017-06-07:
Yeah, I think in general getting a dtb and the driver added into `meta-raspberrypi` would the most user friendly way of adding support for the hardware.

wngr, 2018-02-23:
Any progress on that front?