balena-os/balena-os-device-support #55: Nanopi neo 2

shaunmulligan, 2017-07-13:
CPU: Allwinner H5, Quad-core 64-bit high-performance Cortex A53
DDR3 RAM: 512MB
Connectivity: 10/100/1000M Ethernet, RTL8211E-VB-CG chip
USB Host: USB Type A x 1 and USB pin header x 2
MicroSD Slot: MicroSD x 1 for system boot and storage
LED: Power LED x 1, System LED(Blue) x 1
GPIO1: 2.54mm pitch 24 pin-header, compatible with Raspberry Pi's GPIO pin1 - pin 24. It includes UART, SPI, I2C, IO etc
GPIO2: 2.54mm pitch 12 pin-header. It includes USB, IR receiver, I2S, IO etc
Serial Debug Port: 2.54mm pitch 4pin-header
Audio In/Out: 2.0mm pitch 5 pin-header
MicroUSB: Power input(5V/2A) and OTG
PCB Dimension: 40 x 40mm
Weight: 13g(WITHOUT Pin-headers)

**Link:** http://www.friendlyarm.com/index.php?route=product/product&product_id=180

rodcul, 2018-03-15:
+1 from me, looking to run something like https://www.smallnetbuilder.com/wireless/wireless-howto/33187-build-a-wi-fi-performance-analyzer-for-75-part-1

spodskubka, 2020-03-22:
Support for this, and the newer NanoPi Neo 2 Black (which is roughly the same, but noteable changes include more RAM + eMMC socket) would be incredible!

Especially since this little package provides you with an inexpensive, readily available, off-the-shelf solution for an IoT device with user feedback (small OLED display) and user input (three customizable buttons):
https://www.friendlyarm.com/index.php?route=product/product&product_id=276
[image]

The only thing missing is integrated WiFi, but that wouldn't be a problem for my use case.

I don't know, should there be a separate issue for the Neo 2 Black despite it's similarity to the Neo 2?