balena-os/balena-os-device-support #59: Revolution Pi RevPi Core3

bgentry, 2017-10-12:
This device is based on the eMMC Raspberry Pi 3 compute module.

* **CPU**:  Broadcom BCM2837
* **GPU**: -
* **Memory**: 1GB
* **Storage**: 4GB eMMC
* **Features**:
  * [Analog and digital I/O modules](https://revolution.kunbus.com/io-modules/)
  * [Gateway modules](https://revolution.kunbus.com/gateways/) to bridge other industrial control interfaces

* **Link**: https://revolution.kunbus.com/revpi-core/

There was some discussion about this just today on their forum: https://revolution.kunbus.de/forum/viewtopic.php?f=6&t=418&p=1601&hilit=resin#p1601

[image]

pardegerman, 2017-10-12:
It seems that, in order to have this functioning we need a -rt patched kernel and custom drivers from kunbus. They are all available in their github repo over here https://github.com/RevolutionPi

Unfortunately, I now too little of yocto to understand how to add the relevant patches and drivers to the kernel build.

jgentes, 2018-05-03:
We're looking at Kunbus, would be great to see Resin support for it.

simmikolon, 2018-06-25:
For the RevPi Core 3 Module it's no problem to boot a standard Raspberry Pi 3 ResinOS-Image, since this is just a Compute Module 3 with a Base-Board ... We have however tried to patch the Kernel and used an example of the Resin-Io-Playground on how to build Kernel Extensions. 

Problem however is that there is a driver needed for a MAX3x-IC which Kunbus uses for their IO-Modules. Kunbus has provided that driver and it's now part of the regular Linux Kernel. Unfortunately just in 4.15.

See -> https://www.spinics.net/lists/devicetree/msg191469.html

We tried to build a custom Yocto BSP with an updated Kernel. But ResinOS / Yocto somehow is still using 4.9 / 4.14 - Called *Rocko* and *Sumo*

So we have to wait until there is a 4.15 based Yocto!

Once there is - we will build a solution and open-source it ... We'll let you know!

jgentes, 2018-06-26:
Thank you for this update @simmikolon - we will follow up with the folks at Kunbus to see if we can get a 4.15 based version of their OS. We very much appreciate your effort!

ruudboon, 2018-09-11:
Looking at Kunbus also. Would be awesome to have it running on Resin

simmikolon, 2018-09-12:
@jgentes I've seen an official announcement that ResinIO now supports the Kubus modules "officially" - does that mean you have updated your Yocto base system?

jgentes, 2018-09-13:
@simmikolon we're testing the new version now. It has been working fine with the 12.x version of Resin OS, but this new version is expected to provide support for their expanded modules, such as Digital IO.

coalpower, 2018-09-21:
It would be very interesting to have RevPi Core3 with their IO modules supported on Resin OS!!! 
+1

jgentes, 2018-11-14:
For those trying to get it running, it's worth noting:

1. It can't be flashed with a standard USB drive since USB-boot is disabled and they don't plan on changing that (https://revolution.kunbus.de/forum/viewtopic.php?t=468)

2. Once you flash via micro-USB cable, if you don't remove the cable from the device, it won't show up in the Balena dashboard. You must remove the cable and reboot, then it shows up.

simmikolon, 2019-07-05:
> Thank you for this update @simmikolon - we will follow up with the folks at Kunbus to see if we can get a 4.15 based version of their OS. We very much appreciate your effort!

Hey @jgentes - I see the following update here:

@spanceac wrote:
> This board is currently supported and the DIO module works too

@jgentes does that mean that we now have Kernel Version 4.15?

comisso, 2019-10-07:
> This board is currently supported and the DIO module works too

Hey @spanceac have you really been able to deploy a service that can interact with the DIO module? Can you share something about that?