balena-os/balena-os-device-support #61: NanoPi Neo Plus2

antskelton, 2017-11-30:
SoC: Allwinner H5，Quad-core 64-bit high-performance Cortex A53
DDR3 RAM:1GB
Storage: 8GB eMMC
Network: Gbps Ethernet
WiFi: 802.11b/g/n
Bluetooth: 4.0 dual mode
USB Host: 2 x Independent USB Host
MicroSD Slot: 1 x Slot. It supports system booting or is used to hold a storage card
Audio Input/Output: 5-Pin, 2.0mm pitch pin-header
MicroUSB: power input
Debug Serial: 4Pin, 2.54mm pitch pin-header
GPIO1:24Pin, 2.54mm pitch double-row pin-header containing UART, SPI, I2C and IO
GPIO2:12Pin, 2.54mm pitch pin-header containing USB, IR receiver,I2S and IO
Power Supply: DC 5V/2A
PCB Dimension: 40 x 52mm
PCB Layer: 6-Layer

http://www.friendlyarm.com/index.php?route=product/product&product_id=196

rench, 2018-03-07:
http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO_Plus2

http://wiki.friendlyarm.com/wiki/index.php/Mainline_U-boot_and_Linux

shawaj, 2018-10-22:
I guess this would add support to Neo Core2 as well?