balena-os/balena-os-device-support #64: NanoPi Duo

iSsO, 2018-01-09:
* **CPU:** Allwinner H2+, Quad-core Cortex-A7
* **DDR3 RAM:** 256MB/512MB
* **Connectivity:** 10/100M Ethernet
* **Wifi:** XR819
* **USB Host:** 2.54mm pin x2, exposed in 2.54mm pitch pin header
* **MicroSD Slot x 1**
* **MicroUSB:** OTG and power input
* **Debug Serial Interface:** exposed in 2.54mm pitch pin header
* **Audio input/output Interface:** exposed in 2.54mm pitch pin header
* **GPIO:** 2.54mm spacing 12pin x2. It includes UART, SPI, I2C, IO etc
* **PCB Dimension:** 25.4 x 50mm
* **Power Supply:** DC 5V/2A

* **Link**: http://www.friendlyarm.com/index.php?route=product/product&product_id=197

ryanteck, 2018-03-19:
+1
Also from what I can tell almost identical to the Orange Pi Zero and has armbian images.

mwohlert, 2019-02-17:
I created a pr for the Orange Pi Zero: balena-os/balena-allwinner#65
Maybe you can take this as a refence point and add support for the NanoPi Duo.