balena-os/balena-os-device-support #71: support for AGX Xavier

gancev, 2018-12-23:
* **CPU**:  8-core ARM v8.2 64-bit CPU, 8MB L2 + 4MB L3
* **GPU**:512-core Volta GPU with Tensor Cores
* **Memory**:16GB 256-Bit LPDDR4x | 137GB/s
* **Storage**:
* **Features**:
  *
  *
  *

* **Link**:
https://developer.nvidia.com/embedded/buy/jetson-agx-xavier-devkit

samiamlabs, 2019-04-16:
Hi,

I'm trying to build https://github.com/acostach/jetson-xavier

When I run ```git submodule update --init --recursive``` I get:
```Cloning into 'layers/meta-tegra'...
remote: Enumerating objects: 23, done.
remote: Counting objects: 100% (23/23), done.
remote: Compressing objects: 100% (19/19), done.
remote: Total 6515 (delta 9), reused 11 (delta 4), pack-reused 6492
Receiving objects: 100% (6515/6515), 1.15 MiB | 693.00 KiB/s, done.
Resolving deltas: 100% (3769/3769), done.
Checking connectivity... done.
fatal: reference is not a tree: 379f281880d61ed04da738b143e85a84009d6f8f
```
The folder for ```meta-tegra``` remains empty and the commit does not appear to exsist.

What is the status of this?

acostach, 2019-04-16:
Hello @samiamlabs,

The repository you mentioned is not a BalenaOS maintained one. Jetson Xavier support might be added in the future in BalenaOS, as longer development is necessary for this type of board. Please stay tuned to the blogs at https://www.balena.io/blog/ for device release updates.

samiamlabs, 2019-04-16:
Ok, the links in the readme.md made me think it was. Thank you for answering.