balena-os/balena-os-device-support #78: Nano Pi Neo Core & Core 2

ryanteck, 2019-03-19:
* **CPU**:   Allwinner H3 at up to 1.2 Ghz or Allwinner H5
* **Memory**: 256MB / 512MB / 1GB depending on version
* **Storage**: 8GB / 16GB / 32GB eMMC Onboard
* **Features**:
  *  10/100/1000M Ethernet (1000m is Core 2 Only)
  * Compact Form Factor

* **Link**:http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO_Core#Hardware_Spec  & http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO_Core2#Hardware_Spec

Should be similar / identical to https://github.com/balena-os/resin-os-device-support/issues/38 except for boot files.

The Neo Core is the same as the Neo Air except in different form factor. The Neo Core 2 is the same as the Neo Air 2

floion, 2019-03-23:
@shaunmulligan do we have this hardware around somewhere?

shaunmulligan, 2019-03-25:
@floion I don't think anyone in the team has this hardware yet, unless you guys have them somewhere?