balena-os/balena-os-device-support #79: Sapphire FS-FP5V

arafey, 2019-03-21:
* **CPU**:  
AMD Ryzen™ Embedded V1000 Series (V1807B, V1756B, V1605B, or V1202B)

* **GPU**:
AMD Radeon™ Vega (11, 8, or 3) Graphics

* **Memory**:
2x DDR4-3200 ECC SO-DIMMs

* **Storage**:
  * 1x SATA III
  * M.2 M Key slot (PCIe x 4 or SATA III)

* **Features**:
_see links below_

* **Link**:
[Official Product Page](http://sapphiretech.com/en/commercial/amd-fs-fp5v)
[Detailed Specifications](https://github.com/balena-os/resin-os-device-support/files/2993621/FS-FP5V.user.manual.20190315.pdf)

arafey, 2019-03-21:
I have the hardware.

floion, 2019-03-23:
Did you try to use the Intel NUC OS image on this hardware?

arafey, 2019-03-25:
I will try that today and report on the results.

arafey, 2019-03-27:
It appears to work! I haven't yet looked into all the relevant kernel modules but I suspect this will be a good starting point for making something more specific. My biggest concern is getting the GPU to work because there were issues with the Arch Linux kernel (I suspect it was lacking Vega firmware blobs maybe?).

steffenmllr, 2019-12-11:
@arafey did you made any progress on this. I have a [udoo bolt](https://www.udoo.org/udoo-bolt/) and was wondering on how to get the gpu working. Flashing the NIC OS Image worked fine.