balena-os/balena-os-device-support #82: Coral Dev Board

agherzan, 2019-05-03:
* **CPU**:  NXP i.MX 8M SOC (quad Cortex-A53, Cortex-M4F)
* **GPU**: Integrated GC7000 Lite Graphics
* **Memory**: 1 GB LPDDR4
* **Storage**: 8 GB eMMC 
* **Features**:
  * Wi-Fi 2x2 MIMO (802.11b/g/n/ac 2.4/5GHz) Bluetooth 4.1 
  * Google Edge TPU coprocessor
  *

* **Link**: https://coral.withgoogle.com/products/dev-board/

harborretail, 2019-07-19:
Any update available on this development?

Very interested in this hardware platform.

Jaeyoung-Lim, 2019-09-27:
Would be pretty cool to have this in the fleet!

slessans, 2019-11-12:
Any updates on this?

acostach, 2019-11-27:
yes @slessans , it's in PR stage, will most likely become available in the staging dashboard in the following days: https://github.com/balena-os/balena-coral/pull/1